orchestracities/lego-diorama-usi-2022 #9: Turn on streetlight (raspberrypi)

chicco785, 2022-09-23:
- monitor mqtt queue (1 topic for streetlight or for street)
- if message received turn on light

Solargi, 2022-10-07:
Done the street_lights_client is configured with the config.ini file. when the message "turn on light" is received on the corresponding topic wither the lights of street 1 or 2 will be turned on for 5 seconds, the client will also publish the message "light on" on the topic that sent the first message. (the files are in the traffic directory)

chicco785, 2022-10-08:
> tested the led module by connecting the actual streetlights and it works

the lego ones?

Solargi, 2022-10-09:
yes

inchesg, 2022-10-13:
to verify MQTT format see milestone #18 and #19, this task can be considered closed @chicco785

closed 2022-10-20 by chicco785